dvr0/bzzzbz #1: Sound processing design

daviderovell0, 2020-01-23:
We need to design specifications for the audio processing of bzzzbz video synth.

Currently ADC chip with jack input seems the most reasonable option.

deetrone, 2020-01-23:
E-mail Dr Porr to see if we can use external sound cards (hats)
Research available hats (important 192kHz, price, in/out?)

deetrone, 2020-01-24:
OG I2S specification:
https://web.archive.org/web/20080706121949/http://www.nxp.com/acrobat_download/various/I2SBUS.pdf

Pi as an oscilloscope (Kernel building stuff):
https://digibird1.wordpress.com/raspberry-pi-as-an-oscilloscope-10-msps/

Sticky thread on RPi forum re: I2S:
https://www.raspberrypi.org/forums/viewtopic.php?f=44&t=8496&start=575#p604231

deetrone, 2020-01-25:
I2S GPIO connections to TI PCM1808 ADC:

BCK/SCK - PCM-CLK-18 - pin8 on ADC (BCK)  

WS/LRCLK/FS - PCM-FS 19 -  pin7 on ADC (LRCK)

SDIN - PCMDIN 20 - pin9 on ADC (DOUT)

+ clock, SCKI on ADC which is the "fast clock"

closed 2020-04-02 by daviderovell0