dvr0/bzzzbz #14: Raspbian GUI bypass

daviderovell0, 2020-03-04:
(#2)

It seems possible to bypass the Raspbian GUI and display the video directly on the monitor, without going through the OS.
This could improve latency and make the device more portable by "hiding" the OS to the user. 

This guy seems to be doing it [here](https://www.youtube.com/watch?v=82PHzktgSNA&feature=youtu.be ).

daviderovell0, 2020-04-09:
Implemented using X11 server empty window from CLI interface and filling it through OpenGL main code calling glutWindowInit. 
It is done by adding the compiled graphics code binary into .xinitrc. This file is executed by the `startx` command that opens a window that can be filled by the OpenGL buffer.
This task turned out to be quite challenging as the interaction between OpenGL and underlying windowing systems is complex and can be implemented in several ways. We should write a wiki page that covers these possibilities, describes our approach and add an entry on how to run bzzzbz GUI-less in the execution guide.

deetrone, 2020-04-19:
Solved for Release 1.0

closed 2020-04-19 by deetrone