dvr0/bzzzbz #53: CV and CLK in

deetrone, 2020-04-30:
Implement 4 (?) CV in and 1 clock in. With 4 knobs a single MCP3008 ADC might be enough but in the end might need two of them.

daviderovell0, 2020-09-02:
CV and some gates can exceed 10V up to 12V (?), so the MCP3008 is not enough as it has a max input of 7.6V. We have to use another ADC or decrease the voltage though some additional circuits

marcellillyes, 2020-09-02:
Yess I think we should first try out adding a simple op amp circuit to map
the eurorack range to MCP range.

daviderovell0 <notifications@github.com> ezt írta (időpont: 2020. szept.
2., Sze, 16:24):

marcellillyes, 2020-09-02:
Ill try model it in LTSpice

Marcell Illyés <300000persec@gmail.com> ezt írta (időpont: 2020. szept. 2.,
Sze, 16:33):

daviderovell0, 2020-09-02:
Sounds good. We can check also ADCs that take higher voltages so we can avoid making extra circuit