dvr0/bzzzbz #62: terminate called after throwing an instance of 'char const*'

henning, 2020-12-31:
Hi, 

I dont get the program to work (on an Ubuntu 20.10):

```
$ ./src/bz 
terminate called after throwing an instance of 'char const*'
Aborted (core dumped)
``` 

I never really learned C++, so before spending hours debugging, I ask if this is a known thing, or if there is a recommended Linux version where it should work, or if I better only try with a Raspberry device. 

Thanks in advance! The project looks very promising and I would love to be able to use it! ;)

deetrone, 2021-01-02:
Hi, 

The executable you are trying to run only works on Raspberry Pi with the control interface built. On a desktop device you can run some of the examples instead. There is more information in the user guide of our [wiki](https://github.com/daviderovell0/bzzzbz/wiki/User-Guide).

Great to hear your interest in the project :)

closed 2021-01-13 by deetrone

henning, 2021-01-14:
ok, thanks and sorry for not reading the docs properly - my bad :(

deetrone, 2021-01-18:
@henning No worries. We're glad you're giving our project a try even though it's in its infancy. Let us know if you have other issues or ideas. Admittedly the release is very early and rather unstable. We're hoping to get ahead with development and have a new version that is truly usable and fun so keep an eye on bzzzbz :)